usbarmory/tamago #4: TamaGo fails to initialize on Raspberry Pi 2 (v1.1)

kenbell, 2020-07-29:
I've raised a PR to drop out of HYP mode (default on modern Pi firmware) against tamago-go: https://github.com/f-secure-foundry/tamago-go/pull/1

With this, and some basic board support for Pi2 I'm able to get a hello world / blinky lights app working.

abarisani, 2020-07-30:
Thanks, we'll take a look! Any chance of getting the other tamago patches to get Pi2 support published? We would love to integrate those as well.

kenbell, 2020-07-30:
Absolutely - working on tidying them up.  Will raise a PR.

kenbell, 2020-08-03:
@abarisani - thanks for merging the HYP mode change.  I've got a couple more PRs to add support for Pi 2B and Pi Zero - one for tamago-go and one for tamago itself:
https://github.com/f-secure-foundry/tamago-go/pull/2
https://github.com/f-secure-foundry/tamago/pull/5

Combined these support UART, Watchdog, on-board LEDs.

I've also got a couple examples - one for each board:
https://github.com/kenbell/tamago-example-pizero
https://github.com/kenbell/tamago-example-pi2

If you're interested in hosting this code in this project, let me know - happy to contribute them.

closed 2020-08-04 by abarisani