usbarmory/tamago #8: Raspberry Pi use and the need for U-boot

prusnak, 2020-09-22:
Section `Executing and debugging` of https://github.com/f-secure-foundry/tamago/blob/master/board/raspberrypi/README.md mentioned U-Boot, but it seems that both examples ([Pi Zero](https://github.com/kenbell/tamago-example-pizero) and [Pi 2](https://github.com/kenbell/tamago-example-pi2)) got rid of the U-boot recently.

Can you please clarify the situation and if the U-Boot is really no longer needed update the README.md to not mention it?

abarisani, 2020-09-23:
@kenbell can you comment?

kenbell, 2020-09-23:
Oh yeah - we should update README.  I'll work on an update.

You can either follow the docs and do U-boot, or do as the examples and prepend a stub to the binaries to emulate the linux kernel to the Pi firmware.  Either should work.  Avoiding U-boot is a bit quicker to load.

bburky, 2020-09-23:
Please expand the README to include what files are needed from "an existing Raspberry Pi bootable SD card (note critical Pi firmware is loaded from the SDcard, so blank cards cannot be used)." That just refers to needing an existing `/boot` partition, right?

kenbell, 2020-09-24:
@bburky - yes in raspbian, it's /boot, on the micro-sd card itself it's the files in the root directory.

abarisani, 2020-09-25:
This should be fixed now that https://github.com/f-secure-foundry/tamago/pull/11 has been merged, if there are more concerns please re-open.

Thanks!

PR #11 (2020-09-25): Fix pi readme
Fix #8 - improve Pi readme
closed 2020-09-25 by abarisani