usbarmory/tamago #15: Feasibility of using brutella/hc with tamago

hazcod, 2020-09-28:
Hi, I am looking into writing a HomeKit serial controller for my HVAC system.
This would mean I'd use a microcontroller-like device (such as the Pi zero) to talk serial to the HVAC, and present itself as a HomeKit device over ethernet. 

However I read that tamago does not support threads. Would this mean any project using concurrency is automatically Out of Scope? (e.g. using `go func`)

https://github.com/brutella/hc

abarisani, 2020-09-28:
Threads != concurrency.

Go routines work just fine in TamaGo and so is concurrent programming.

As long as the imported packages honor [these limitations](https://github.com/f-secure-foundry/tamago/wiki/Internals#go-application-limitations) then you are good to go.

The lack of threads does not prevent using go routines and concurrency.

Please also note that the Pi Zero is an SoC based device and not an MCU (microcontroller) device. MCU like devices are suited for TinyGo rather than TamaGo. Also note that currently Pi drivers do not support USB (and therefore Ethernet).

closed 2020-09-28 by abarisani

hazcod, 2020-09-28:
Ahh, shucks. Should I rename this ticket to 'Pi USB driver support' for tracking?

abarisani, 2020-09-28:
It's a clear need, not sure about the timeline. If you want to raise an issue feel free to do it but regardless of that it's a need that won't be ignored ;).

Cheers

hazcod, 2020-12-15:
@abarisani I just rediscovered this issue, can we re-open since it's renamed for USB/ethernet support?

abarisani, 2020-12-15:
We have no plans to write USB support for the Pi ourselves, however if and when an external contribution is made we would welcome it and be happy to merge it.