usbarmory/tamago #18: error typechecking package os/signal: /home/saper/src/tamago-go/src/os/signal/signal.go:32:9: undeclared name: numSig (tamago-arm)

saper, 2020-11-08:
Tried building tamago-go (c117e5d62adf00b99dc5cb9e7e0d3105d87fb09d - current `latest`) on FreeBSD 11.4 and got this test failure in the end:

##### API check
Error running API checker: exit status 1
Go version is "go1.13.4", ignoring -next /home/saper/src/tamago-go/api/next.txt
2020/11/08 18:32:44 error typechecking package os/signal: /home/saper/src/tamago-go/src/os/signal/signal.go:32:9: undeclared name: numSig (tamago-arm)
exit status 1
2020/11/08 18:36:14 Failed: exit status 1
2020/11/08 18:36:14 FAILED

go-1.15.2,1 from FreeBSD standard packages has been used to bootstrap it.

Full build log: https://gist.github.com/be54c7dfd7a45b7c611a9999f240c528

abarisani, 2020-11-08:
There is a problem with the `latest` tag, for now please checkout `tamago1.15.3`.

saper, 2020-11-08:
It works with tamago1.15.3, thanks.

abarisani, 2020-11-09:
The `latest` tag has now been fixed, thanks for spotting this.

closed 2020-11-09 by abarisani