usbarmory/tamago #19: relocation target runtime.nanotime not defined

saper, 2020-11-08:
I try to build armory-ums on FreeBSD 11.4 using tamago compiled there (with one test failing at https://github.com/f-secure-foundry/tamago/issues/18):

```
saper@guest4:~/src/armory-ums % gmake TAMAGO=$HOME/src/tamago-go/bin/go SHELL=/usr/local/bin/bash
GO_EXTLINK_ENABLED=0 CGO_ENABLED=0 GOOS=tamago GOARM=7 GOARCH=arm /home/saper/src/tamago-go/bin/go build -tags linkramsize -ldflags "-s -w -T 0x80010000  -E _rt0_arm_tamago -R 0x1000 -X 'main.Build=saper@guest4 on 2020-11-08 19:07:20' -X 'main.Revision=39e2ed4'" -o armory-ums
# github.com/f-secure-foundry/armory-ums
runtime.chansend: relocation target runtime.nanotime not defined
runtime.send: relocation target runtime.nanotime not defined
runtime.closechan: relocation target runtime.nanotime not defined
runtime.chanrecv: relocation target runtime.nanotime not defined
runtime.recv: relocation target runtime.nanotime not defined
runtime.notetsleepg: relocation target runtime.nanotime not defined
runtime.checkTimeouts: relocation target runtime.nanotime not defined
runtime.(*gcControllerState).endCycle: relocation target runtime.nanotime not defined
runtime.(*gcControllerState).findRunnableGCWorker: relocation target runtime.nanotime not defined
runtime.pollFractionalWorkerExit: relocation target runtime.nanotime not defined
runtime.gcStart: relocation target runtime.nanotime not defined
runtime.gcMarkDone: relocation target runtime.nanotime not defined
runtime.gcMarkTermination: relocation target runtime.nanotime not defined
runtime.gcBgMarkWorker: relocation target runtime.nanotime not defined
runtime.gcAssistAlloc1: relocation target runtime.nanotime not defined
runtime.gcPaceScavenger: relocation target runtime.nanotime not defined
runtime.scavengeSleep: relocation target runtime.nanotime not defined
runtime.usleep: relocation target runtime.nanotime not defined
runtime.main: relocation target runtime.nanotime not defined
runtime.forcegchelper: relocation target runtime.nanotime not defined
runtime.schedinit: relocation target runtime.nanotime not defined
/home/saper/src/tamago-go/pkg/tool/freebsd_amd64/link: too many errors
gmake: *** [Makefile:68: armory-ums] Error 2
saper@guest4:~/src/armory-ums % git log -1
commit 39e2ed4306c66341feefe39ed925be7c10506a6c (HEAD -> master, tag: v20201102, origin/master, origin/HEAD)
Author: Andrea Barisani <andrea@inversepath.com>
Date:   Mon Nov 2 12:27:46 2020 +0100

    fix REPORT LUNS response format
```

abarisani, 2020-11-08:
Did you checkout the `latest` branch or a tamago release branch?

saper, 2020-11-08:
> Did you checkout the `latest` branch or a tamago release branch?

it's `latest`...

abarisani, 2020-11-08:
There is a problem with the `latest` tag, for now please checkout `tamago1.15.3`.

saper, 2020-11-08:
Yes, I have managed to build it with `tamago1.15.3` and it worked fine. The produced armory-usb.imx image boots fine, thanks!

abarisani, 2020-11-09:
The `latest` tag has now been fixed, thanks for spotting this.

closed 2020-11-09 by abarisani