usbarmory/tamago #23: New compilation error after 046dc6d

jphastings, 2021-04-28:
Hi folks,

A [recent commit](https://github.com/f-secure-foundry/tamago/commit/046dc6d13c937abbe3d71e4ccab7a3807b817c11) has prevented Tamago from compiling code for my Raspberry Pi. `arm/exception.go` and `arm/mmu.go` both now refer to a `runtime.MemRegion()` function ([here](https://github.com/f-secure-foundry/tamago/blob/master/arm/exception.go#L134), [here](https://github.com/f-secure-foundry/tamago/blob/master/arm/mmu.go#L60)), which doesn't appear to exist.

(I'm compiling the [pi zero example](https://github.com/kenbell/tamago-example-pizero), having replaced the `go.mod` file, to ensure the module references are appropriate)

Am I missing something?

```
$ echo $TAMAGO
/usr/local/tamago-go/bin/go
$ /usr/local/tamago-go/bin/go version
go version go1.16.3 linux/amd64

$ /usr/local/tamago-go/bin/go build -ldflags "-s -w -T0x00010000 -E _rt0_arm_tamago -R 0x1000" -o "build/kernel" "./"
# github.com/f-secure-foundry/tamago/arm
/root/go/pkg/mod/github.com/f-secure-foundry/tamago@v0.0.0-20210427101945-af9f214e92ad/arm/exception.go:134:17: undefined: runtime.MemRegion
/root/go/pkg/mod/github.com/f-secure-foundry/tamago@v0.0.0-20210427101945-af9f214e92ad/arm/exception.go:154:17: undefined: runtime.MemRegion
/root/go/pkg/mod/github.com/f-secure-foundry/tamago@v0.0.0-20210427101945-af9f214e92ad/arm/mmu.go:60:17: undefined: runtime.MemRegion
/root/go/pkg/mod/github.com/f-secure-foundry/tamago@v0.0.0-20210427101945-af9f214e92ad/arm/mmu.go:87:16: undefined: runtime.MemRegion
```

abarisani, 2021-04-28:
Recent commits requires tamago-go compiled from https://github.com/f-secure-foundry/tamago-go/commit/cabc6f79d791fb685a0bd18cceea3b0394084b2f or later.

Either use an older tamago revision or update your tamago-go compiler.

We don't have a binary release yet for tamago-go with these changes, I will make one later this week.

closed 2021-04-28 by abarisani

jphastings, 2021-04-28:
Ahh thank you for the tip @abarisani! I'll build from source and update my [tamago-build](https://github.com/jphastings/tamago-build) when you release a new version of tamago-go ☺️