usbarmory/tamago #24: Unable to get example working for RPi Zero

jphastings, 2021-04-29:
Hi folks,

I've had an [issue open on the RPi Zero example repo](https://github.com/kenbell/tamago-example-pizero/issues/2) for a month, but realised that may not be the right place to post this problem!

Building the example RPi Zero project ([with some tweaks, to fix the go.mod file](https://github.com/jphastings/tamago-example-pizero/tree/d478bd44f6462df42ff87adac1cbd9078798d21a)) creates a kernel file which builds successfully, but fails to run on any of my RPi Zeros.

[These are the files](https://github.com/f-secure-foundry/tamago/files/6396993/tamago-rpi0.zip) I've had on my SD card, but I get no flashing LED, no printed output and no clues I know how to follow as to further debugging (I also tried including `bcm2708-rpi-zero-w.dtb` as it's called out as missing in the UART log, with no luck).

How should I proceed in trying to fix this?

<details>
  <summary>UART Output</summary>

```
Raspberry Pi Bootcode

Read File: config.txt, 279

Read File: start.elf, 2955648 (bytes)

Read File: fixup.dat, 7311 (bytes)

MESS:00:00:01.002000:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.006316:0: brfs: File read: 279 bytes
MESS:00:00:01.062589:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.068754:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.075090:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.081427:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.087764:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.094100:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.100437:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.106774:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.113111:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.119448:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.125543:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.131252:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.136292:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.163018:0: brfs: File read: 279 bytes
MESS:00:00:01.343494:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:01.349317:0: *** Restart logging
MESS:00:00:01.353713:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.361298:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.368156:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.375014:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.381870:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.388729:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.395586:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.402443:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.409302:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.416158:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.422775:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.428679:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.436472:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.443329:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.450187:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.457044:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.463902:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.470760:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.477617:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.484475:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.491333:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.497949:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.503563:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:01.512308:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:01.518005:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
MESS:00:00:01.539405:0: dtb_file 'bcm2708-rpi-zero-w.dtb'
MESS:00:00:01.543299:0: dtb_file 'bcm2708-rpi-0-w.dtb'
MESS:00:00:01.548141:0: Failed to load Device Tree file 'bcm2708-rpi-0-w.dtb'
MESS:00:00:01.555008:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:01.719713:0: brfs: File read: /mfs/sd/example-pi-zero.bin
MESS:00:00:01.724374:0: Loading 'example-pi-zero.bin' to 0x8000 size 0x161050
MESS:00:00:01.731301:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:01.739720:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:01.746151:0: uart: Baud rate change done...
MESS:00:00:01.749584:0: uart: Baud ratel1pageTableStart 0x4000
l1pageTableSize  0x4000
```

</details>

abarisani, 2021-04-29:
We will try to reproduce as time allows (Pi support is not our primary focus and the PR on kenbell repository certainly is reaching the right person for this). Nonetheless we will try to take a look and get back to you.

In the meantime please make sure that tamago-go and tamago are up to the latest version (the UART output suggests an older version).

Thanks

jphastings, 2021-04-29:
Thank you @abarisani; I recognise the RPi isn't a core part of your work and really appreciate the time you've taken! I'm interested in contrinuting and helping to fix this (both for myself, and for others) — if you've even any 'gut' instincts they'd go a long way to helping me, as my professional experience is on the web stack, not kernel!

I'll keep exploring, and trying to reach @kenbell for advice, but in the meanwhile (for reference), I'll post any advances in this thread.

---

Just to ensure this wasn't a versioning issue, I made sure I was using commit [9f55bc1](https://github.com/f-secure-foundry/tamago/tree/9f55bc15675283e5d3f30b420d26e9dbd6ec67ce) — the most recent that works with the current 1.16.3 `latest.zip` pre-build tamago-go. I also added `dtdebug=1` in `config.txt`, but I can't see anything else useful in there.

I've been exploring the RPi's [UART configuration](https://www.raspberrypi.org/documentation/configuration/uart.md), as the final lines appear slightly garbled, and they're just after a log line declaring a resetting of the UART baudrate; but with no luck.

Tips, from anyone, much appreciated!

<details>
  <summary>UART Output (Recent build + dtdebug=1)</summary>

```
Raspberry Pi Bootcode
Read File: config.txt, 289
Read File: start.elf, 2955968 (bytes)
Read File: fixup.dat, 7311 (bytes)
MESS:00:00:01.328287:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.332628:0: brfs: File read: 289 bytes
MESS:00:00:01.389114:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.395281:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.401617:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.407954:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.414291:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.420628:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.426964:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.433301:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.439638:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.445975:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.452071:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.459525:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.463810:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.278534:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.284358:0: *** Restart logging
MESS:00:00:02.288236:0: brfs: File read: 289 bytes
MESS:00:00:02.293236:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.300852:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.307711:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.314568:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.321426:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.328283:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.335141:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.341999:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.348857:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.355714:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.362330:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:02.368235:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.376026:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.382885:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.389742:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.396600:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.403457:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.410315:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.417173:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.424030:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.430888:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.437504:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:02.443118:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:02.451863:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:02.457569:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
MESS:00:00:02.479003:0: dtb_file 'bcm2708-rpi-zero-w.dtb'
MESS:00:00:02.482701:0: Trying Device Tree file 'bcm2708-rpi-zero-w.dtb'
MESS:00:00:02.493751:0: brfs: File read: /mfs/sd/bcm2708-rpi-zero-w.dtb
MESS:00:00:02.498670:0: Loading 'bcm2708-rpi-zero-w.dtb' to 0x100 size 0x67b1
MESS:00:00:02.515064:0: dtdebug: delete_node(/__local_fixups__)
MESS:00:00:02.522228:0: brfs: File read: 26545 bytes
MESS:00:00:02.525654:0: dtdebug: using platform 'bcm2835'
[... 19 lines cut ...]
MESS:00:00:02.679115:0: dtdebug:   override uart0: string target 'status'
MESS:00:00:02.695551:0: brfs: File read: 289 bytes
MESS:00:00:02.698734:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:02.704019:0: Failed to read command line file 'cmdline.txt'
MESS:00:00:02.722622:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:04.928760:0: dtparam: cam0-pwdn=44
[... cut ...]
MESS:00:00:05.083061:0: brfs: File read: /mfs/sd/example-pi-zero.bin
MESS:00:00:05.087723:0: Loading 'example-pi-zero.bin' to 0x8000 size 0x161050
MESS:00:00:05.094589:0: Device tree loaded to 0x1bfe9400 (size 0x6ba3)
MESS:00:00:05.101887:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:05.108550:0: uart: Baud rate change done...
MESS:00:00:05.111981:0: uart: Baud ratel1pageTableStart 0x4000
                                                              l1pageTableSize  0x4000
```

</details>

abarisani, 2021-05-08:
Can you please try with tamago HEAD and our [latest tamago-go binary release](https://github.com/f-secure-foundry/tamago-go/releases)? Thank you.

jphastings, 2021-05-18:
Hey @abarisani, I'm not getting past the build step:

```
$ docker run -v $(pwd):/workdir jphastings/tamago-build:v1.16.4
/root/go/pkg/mod/github.com/f-secure-foundry/tamago@v0.0.0-20210512135525-47ab2cac2668/soc/bcm2835/bcm2835.go:58:5: ARM.CacheEnable undefined (type *arm.CPU has no field or method CacheEnable)
```

You can see I'm using the latest commit of tamago ([47ab2cac2668](https://github.com/f-secure-foundry/tamago/commit/47ab2cac266889d0302e77b942a18a1b568a872e)), run with [the tamago 1.16.4 binary build](https://github.com/f-secure-foundry/tamago-go/releases/tag/tamago-go1.16.4). I've not delved into why this error might be occurring yet, not have I (yet) tried using the source-built tamago-go.

I'll take a look into this, but it's likely beyond my knowlegde!

abarisani, 2021-05-18:
Fixed in https://github.com/f-secure-foundry/tamago/commit/ddf03422d158c43a773ae431044557aeb2eb0040

Can you try with it?

jphastings, 2021-05-18:
It successfully builds, but never appears to execute the application code in a way similar (but not identical) to above.

<details>
  <summary>UART Output (ddf0342 build)</summary>

```
Raspberry Pi Bootcode
Read File: config.txt, 277
Read File: start.elf, 2955936 (bytes)
Read File: fixup.dat, 7313 (bytes)
MESS:00:00:01.121071:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.125396:0: brfs: File read: 277 bytes
MESS:00:00:01.181688:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.187853:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.194191:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.200527:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.206864:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.213201:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.219538:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.225874:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.232211:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.238548:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.244643:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.252148:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.256417:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.071024:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.076847:0: *** Restart logging
MESS:00:00:02.080726:0: brfs: File read: 277 bytes
MESS:00:00:02.085744:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.093342:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.100200:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.107058:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.113915:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.120773:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.127631:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.134489:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.141346:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.148203:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.154819:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:02.160724:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.168516:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.175374:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.182232:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.189088:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.195947:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.202805:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.209662:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.216520:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.223377:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.229993:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:02.235608:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:02.244353:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:02.250050:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
MESS:00:00:02.271494:0: dtb_file 'bcm2708-rpi-zero-w.dtb'
MESS:00:00:02.279805:0: brfs: File read: /mfs/sd/bcm2708-rpi-zero-w.dtb
MESS:00:00:02.284736:0: Loading 'bcm2708-rpi-zero-w.dtb' to 0x100 size 0x67b1
MESS:00:00:02.304115:0: brfs: File read: 26545 bytes
MESS:00:00:02.374111:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:02.378040:0: dtparam: uart0=on
MESS:00:00:02.394343:0: brfs: File read: 277 bytes
MESS:00:00:02.397525:0: brfs: File read: /mfs/sd/cmdline.txt
MESS:00:00:02.402812:0: Failed to read command line file 'cmdline.txt'
MESS:00:00:02.421429:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:04.752353:0: brfs: File read: /mfs/sd/tamago-kernel.bin
MESS:00:00:04.756840:0: Loading 'tamago-kernel.bin' to 0x8000 size 0x161030
MESS:00:00:04.763533:0: Device tree loaded to 0x1bfe9400 (size 0x6ba3)
MESS:00:00:04.770797:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:04.777492:0: uart: Baud rate change done...
MESS:00:00:04.780936:0: uart: Baud rate
```

</details>

Now I'm getting no output after the Baud rate change I'm worried I'm missing further debug info because it's switching rates.

I've been monitoring the output with a separate serial USB device, and `screen /dev/cu.usbserial-0001 115200`, and the `config.txt` set to:

```txt
enable_uart=1
uart_2ndstage=1
dtparam=uart0=on
kernel_address=0x8000
disable_commandline_tags=1
core_freq=250
kernel=tamago-kernel.bin
```

I realise tis is Pi-specific stuff, but if you know the lack of further output is expected, or some part of this config isn't helping you debug, just mention it!

abarisani, 2021-05-18:
Thanks, we'll try to reproduce this and get back to you.

jphastings, 2021-05-18:
I've refreshed my memory of [RPi UART usage](https://www.raspberrypi.org/documentation/configuration/uart.md) and tried a number of scenarios, without any change or luck. This includes:

- Removing the `core_freq` and `enable_uart` assertions in config.txt (as they seem mostly related to miniUART, and on the RPi Zero, the PL011 is default/primary anyway)
- Added a `dtoverlay=disable-bt` line in `cmdline.txt` (along with [that overlay](https://github.com/raspberrypi/linux/blob/rpi-5.4.y/arch/arm/boot/dts/overlays/disable-bt-overlay.dts) from the rpi repo, at `overlays/disable-bt-overlay.dts`) — as this ensures that the PL011 is primary (this looks to be already the case, from the logs above, but worth a try!)

Thank you for your help!

abarisani, 2021-05-23:
FYI we have identified the likely cause and we are working on a solution, should be done next week.

jphastings, 2021-05-31:
Thank you! I'm looking forward to it 😊

andrejro, 2021-06-01:
The issue has been solved now. Please try our fork: https://github.com/f-secure-foundry/tamago-example-pizero

jphastings, 2021-06-01:
Your fork seems to be using an old version of tamago (maybe you didn't push your changes to your fork?) so I did the following, but still have no luck:

```bash
$ git clone git@github.com:f-secure-foundry/tamago-example-pizero.git

# Get rid of old go mod file
$ rm go.*

# Use latest tamago
$ go mod init
$ go mod tidy

# Ensure we're using the latest tamago
$ cat go.mod
module github.com/f-secure-foundry/tamago-example-pizero

go 1.16

require github.com/f-secure-foundry/tamago v0.0.0-20210601073428-3d51445fa773

# Build with tamago-go 
$ docker run -v $(pwd):/workdir -v /Volumes/SDCARD:/workdir/build jphastings/tamago-build:HEAD
Downloading RPi firmware: LICENCE.broadcom
Downloading RPi firmware: bootcode.bin
Downloading RPi firmware: fixup.dat
Downloading RPi firmware: start.elf
go: downloading github.com/f-secure-foundry/tamago v0.0.0-20210601073428-3d51445fa773
Volume SDCARD on disk3s1 unmounted
```

(This was built with tamago-go built from source @1.16.4, in [this docker container](https://hub.docker.com/repository/registry-1.docker.io/jphastings/tamago-build/builds/d71587d8-9e6d-42ee-83b5-a6ae0018773f) using [this definition](https://github.com/jphastings/tamago-build/blob/main/Dockerfile.source))

<details>
<summary>Same problem 😔</summary>

```
Raspberry Pi Bootcode
Read File: config.txt, 277
Read File: start.elf, 2956320 (bytes)
Read File: fixup.dat, 7310 (bytes)
MESS:00:00:01.114312:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.118632:0: brfs: File read: 277 bytes
MESS:00:00:01.175649:0: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:01.181815:0: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:01.188151:0: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:01.194488:0: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:01.200825:0: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:01.207162:0: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:01.213498:0: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:01.219836:0: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:01.226172:0: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:01.232509:0: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:01.238604:0: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:01.245991:0: brfs: File read: /mfs/sd/config.txt
MESS:00:00:01.250247:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.065225:0: gpioman: gpioman_get_pin_num: pin LEDS_PWR_OK not defined
MESS:00:00:02.071049:0: *** Restart logging
MESS:00:00:02.074928:0: brfs: File read: 277 bytes
MESS:00:00:02.079881:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.087544:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.094402:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.101259:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.108116:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.114979:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.121832:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.128690:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.135548:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.142405:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.149021:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:02.154924:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 0
MESS:00:00:02.162717:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 1
MESS:00:00:02.169576:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 2
MESS:00:00:02.176433:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 3
MESS:00:00:02.183290:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 4
MESS:00:00:02.190149:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 5
MESS:00:00:02.197006:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 6
MESS:00:00:02.203863:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 7
MESS:00:00:02.210722:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 8
MESS:00:00:02.217579:0: hdmi: HDMI0:EDID error reading EDID block 0 attempt 9
MESS:00:00:02.224195:0: hdmi: HDMI0:EDID giving up on reading EDID block 0
MESS:00:00:02.229809:0: hdmi: HDMI:hdmi_get_state is deprecated, use hdmi_get_display_state instead
MESS:00:00:02.238554:0: HDMI0: hdmi_pixel_encoding: 162000000
MESS:00:00:02.244250:0: vec: vec_middleware_power_on: vec_base: 0x7e806000 rev-id 0x00002708 @ vec: 0x7e806100 @ 0x00000420 enc: 0x7e806060 @ 0x00000220 cgmsae: 0x7e80605c @ 0x00000000
MESS:00:00:02.265655:0: dtb_file 'bcm2708-rpi-zero-w.dtb'
MESS:00:00:02.269480:0: dtb_file 'bcm2708-rpi-0-w.dtb'
MESS:00:00:02.274324:0: Failed to load Device Tree file 'bcm2708-rpi-0-w.dtb'
MESS:00:00:02.281191:0: Failed to open command line file 'cmdline.txt'
MESS:00:00:02.374815:0: brfs: File read: /mfs/sd/tamago-kernel.bin
MESS:00:00:02.379302:0: Loading 'tamago-kernel.bin' to 0x8000 size 0x161028
MESS:00:00:02.386055:0: gpioman: gpioman_get_pin_num: pin EMMC_ENABLE not defined
MESS:00:00:02.394440:0: uart: Set PL011 baud rate to 103448.300000 Hz
MESS:00:00:02.400905:0: uart: Baud rate change done...
MESS:00:00:02.404338:0: uart: Baud rate
```

</details>

Am I missing something?

andrejro, 2021-06-01:
I forgot to push on the repo, sorry. Please retry now.

jphastings, 2021-06-01:
It works beautifully! Thank you for the hard work @andrejro 😊

I have some time (and interest) in continuing to develop [tamago-build](https://github.com/jphastings/tamago-build), to allow mac users (like myself), windows users, or linux users without a cross compile set up to build apps with tamago — if that's something you think your team would be interested in, I'd love a chat to make sure I'd be building in the same direction you are.

Thanks again!

closed 2021-06-01 by jphastings

abarisani, 2021-06-02:
> It works beautifully! Thank you for the hard work @andrejro
>
> I have some time (and interest) in continuing to develop [tamago-build](https://github.com/jphastings/tamago-build), to allow mac users [...]

Sure thing, sounds definitely interesting.

jphastings, 2021-06-04:
[PR added here](https://github.com/f-secure-foundry/tamago-go/pull/3) to cover that (just to join up the threads — I'll await your thoughts & feedback over there)