usbarmory/tamago #31: How does tamago compare to gokrazy?

andig, 2021-06-19:
See https://gokrazy.org/. Both seem to provide bare metal, both are lacking Wifi support. gokrazy works with the stock compiler. 
I have some experience with gokrazy- why would I need tamago and its changes to the runtime?

refs https://github.com/golang/go/issues/46802

gsora, 2021-06-19:
I believe gokrazy still uses the Linux kernel underneath, while tamago produces binaries that can be run bare metal, without the need for another kernel.

abarisani, 2021-06-21:
gokrazy is very different, it is not bare metal and uses an actual OS (Linux) underneath, therefore its implementation and scope are extremely different.

This [comparison](https://github.com/f-secure-foundry/tamago/wiki/Frequently-Asked-Questions-(FAQ)#how-does-tamago-differ-from-pastcurrent-similar-efforts) might give you better perspective with more similar efforts.

closed 2021-06-21 by abarisani