usbarmory/tamago #33: add qemu x86 amd64 support?

Bmixo, 2022-04-02:
Is this requirement possible?

andrejro, 2022-04-04:
There are no plans to add amd64 support in near future. Maybe you should check [eggos](https://github.com/icexin/eggos).

abarisani, 2024-11-23:
Re-opening as this is now a WiP and planned for release.

abarisani, 2024-11-23:
The compiler modifications are tracked in https://github.com/usbarmory/tamago-go/pull/11

prologic, 2024-11-24:
Nice! 🙌

abarisani, 2024-12-05:
As of [tamago-go1.23.4](https://github.com/usbarmory/tamago-go/releases/tag/tamago-go1.23.4) the compiler modifications are now [merged](https://github.com/usbarmory/tamago-go/pull/11) and execution through [testing](https://github.com/usbarmory/tamago-go/blob/tamago1.23.4/src/testing/testing_tamago.go) is [supported](https://github.com/usbarmory/tamago-go/blob/tamago1.23.4/src/testing/testing_tamago_amd64.s) to launch [standard distribution tests](https://github.com/usbarmory/tamago/wiki/Compatibility).

Next up, emulated VM hardware support.

abarisani, 2025-01-06:
The tamago modifications are tracked in https://github.com/usbarmory/tamago/pull/45

abarisani, 2025-01-20:
All PRs have been merged and [tamago-example](https://github.com/usbarmory/tamago-example) has been updated with `TARGET=microvm` with working [VirtIO networking](https://github.com/usbarmory/virtio-net) support.

```
$ ssh 10.0.0.1

...

> info
Runtime ......: go1.23.5 tamago/amd64
RAM ..........: 0x10000000-0x50000000 (1024 MiB)
Board ........: microvm
CPU ..........: 11th Gen Intel(R) Core(TM) i7-1165G7 @ 2.80GHz
```

closed 2025-01-20 by abarisani