usbarmory/tamago #43: USB Host mode support

fabled, 2023-09-23:
I was just able to complete a custom USB device mode support on Armory MK2.

Now I am looking to attach other peripherals to the USB-C receptacle and to communicate to those devices. Also interested in having USB passthrough (that is proxy the traffic between the computer on the USB-C plug and the peripheral connected to USB-C receptacle).

I did not see yet any host controller support code, or did I just not find it? Is support for this planned?

abarisani, 2023-09-24:
There are no plans for host mode at this time.

closed 2023-09-24 by abarisani

mstaack, 2024-01-30:
@abarisani any hints how to get closer to a host mode driver?

even basic midi/serial/storage hosting would be really interesting.

abarisani, 2024-01-30:
> There are no plans for host mode at this time.

This is unchanged.